nidayand/motor-on-roller-blind-ws #5: Possible to add option for clockwise/countclockwise in main page?

olskar, 2018-03-22:
Hello!

Nice project!

Is it possible to add option for clockwise/countclockwise in main page? As of now I need to know what way it should turn before I have access to the main interface.

nidayand, 2018-04-03:
Hi,
I was thinking about that... but I didn't have the energy to create another HTML page. Sorry, don't have the time to do it right now though so I'll keep it as an enhancement request

petrepa, 2018-09-10:
Is there anyhow a way to change the rotation? How can I possibly do a hard reset of a configured unit?

nidayand, 2018-09-12:
yeah, I know, it isn't perfect... What you can do is to uncomment row 272 and 273 and flash it via cable and then revert back and flash again.. (or use esptool.py to clean it completely)

  //clean FS, for testing
  //helper.resetsettings(wifiManager);