nidayand/motor-on-roller-blind-ws #8: Home bridge

lukaszwitkowski, 2018-04-07:
Hello i Want to use mqtt in my homebridge.
Can You tell me what pulling gona be best and hav to configure.

nidayand, 2018-04-16:
HI, I don't understand the question. Please elaborate

lukaszwitkowski, 2018-04-16:
I want to use sonething to comunicate witch mqtt
But that dont work
https://github.com/bonze4/homebridge-mqtt-blinds

lukaszwitkowski, 2018-04-16:
My blind
Send message to mqtt
/raw/esp8266/8959633/out { „set”:51, „position”:51 }
And my tablet send message to mqtt
/raw/esp8266/8959633/out/SET/targetPosition 30
When i change to „in”
/raw/esp8266/8959633/in/SET/targetPosition 30
Nothing work

nidayand, 2018-04-16:
The problem with homebridge-mqtt-blinds is that it is using the topic to control the blind and not the payload - it is a matter of design. So you either have to create new new fork of homebridge-mqtt-blinds or motor-on-roller-blind-we to modify the messages being sent. This repo only has one topic that it listens to and that is the "/raw/esp8266/8959633/in" topic.

lukaszwitkowski, 2018-04-16:
For what you add mqtt? Do you use mqtt to something? Do you have kind of pulling?

nidayand, 2018-04-16:
I personally only use MQTT to control my blinds. I have a wireless Xiaomi button and I use Google Home voice control to steer my blinds

nidayand, 2018-04-16:
Pulling is available - if you mean requesting a status.. Just send '(update)' and it will post the current position

nidayand, 2018-04-16:
I got it. In summary you need to modify the code in one end..

closed 2018-04-16 by nidayand