nidayand/motor-on-roller-blind-ws #11: MQTT connection failed

Glivmo, 2018-04-13:
So, I've used version 1.2.0 for quite some time and decided to update to 1.3.1. I'm using this in conjuction with an MQTT server through hass.io, for which i added username and password support myself in 1.2.0.

After making the update to 1.3.1 the blinds are not functioning correctly. I set the wifi settings via the "BlindsConnectAP" network after which it tries to connect to my MQTT server, but it prints out the following:
`Attempting MQTT connection...failed, rc=-2 try again in 5 seconds`
It then just keeps attempting the connection, but never manages to connect. I've already checked if my MQTT server works properly and it does. I tried almost everything (including different usernames), but it will never connect. When not filling in the mqtt_server, it does connect, but only for websocket (duh).

I even tried going back to 1.2.0, but unfortunately it shows the same error. Any thoughts?

nidayand, 2018-04-16:
Yes, clear the flash as the newer release has more variables that is stored. You need to reset it - e.g. clear it with esptool.py or uncomment 👍 

>   //reset settings - for testing
>   //clean FS, for testing
>   //helper.resetsettings(wifiManager);

Glivmo, 2018-04-16:
Thanks for responding,

I tried clearing the flash with esptool.py already. I sent me back to the BlindsConnectAP stuff which i filled in, but after it showed me the error message. I also just tried the method of uncommenting 

> //helper.resetsettings(wifiManager);

Unfortunately it didn't fix the problem. It still gives the 
`Attempting MQTT connection...failed, rc=-2 try again in 5 seconds` error. Any other idea what it could be.

By the way, on lines 263 to 266 the code says the following:

> WiFiManagerParameter custom_mqtt_server("server", "MQTT server", mqtt_server, 40);
> WiFiManagerParameter custom_mqtt_port("port", "MQTT port", mqtt_port, 6);
> WiFiManagerParameter custom_mqtt_uid("uid", "MQTT username", mqtt_server, 40);
> WiFiManagerParameter custom_mqtt_pwd("pwd", "MQTT password", mqtt_server, 40);

Shouldn't the last two lines be:

> WiFiManagerParameter custom_mqtt_uid("uid", "MQTT username", **mqtt_uid**, 40);
> WiFiManagerParameter custom_mqtt_pwd("pwd", "MQTT password", **mqtt_pwd**, 40);

Glivmo, 2018-04-16:
I tried it with the correct settings before (just tried it again hoping that it would maybe just magically work now) but unfortunately that's not the problem. Could you maybe try if the code with authentication works for you? Thanks in advance

nidayand, 2018-04-16:
Sorry to hear that. I can't promise when I'll have the time to test it unfortunately..

Bennedetto, 2018-09-25:
I accidentally run in the same issue (Serial kept printing : "Attempting MQTT connection...failed, rc=-2 try again in 5 seconds" in an endless loop) and fixed it by providing some SPIFFS.
Maybe this will fix your issue as well?

SoundGoof, 2018-12-11:
@Bennedetto Hi, I have the same problem, do you have PR/gist for your fix?

Bennedetto, 2018-12-30:
@SoundGoof Hi, sorry for the late reply.
This is a configuration issue - nothing you may fix with code change. Depending on your flash size, you should select anything equal or lower to you chips size **and including SPIFFS**. The configuration is saved as a json file in the SPIFFS - thus this should be available. This fixed the issue for me.
[image]

closed 2019-01-22 by nidayand