nidayand/motor-on-roller-blind-ws #13: Error compiling for NodeMCU 1.0 (ESP-12E module)

Kepete, 2018-06-12:
I try to compile and upload it using the Arduino IDE and get an error like this: 

> Arduino: 1.8.4 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"
> 
> Build options changed, rebuilding all
> Archiving built core (caching) in: /var/folders/46/nj9tjpqs7sl5mmtg1jf6bfqm0000gn/T/arduino_cache_487244/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_aa4f9839ca6ef0db29b345caf646b2a9.a
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o:(.text._Z7sendmsg6StringS_+0xc): undefined reference to `NidayandHelper::mqtt_publish(PubSubClient&, String, String)'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o: In function `sendmsg(String, String)':
> /Users/user/Downloads/motor-on-roller-blind-ws-master/motor_on_a_roller_blind-ws/motor_on_a_roller_blind-ws.ino:104: undefined reference to `NidayandHelper::mqtt_publish(PubSubClient&, String, String)'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o:(.text._Z10loadConfigv+0x38): undefined reference to `NidayandHelper::loadconfig()'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o:(.text._Z10loadConfigv+0x3c): undefined reference to `NidayandHelper::getconfig()'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o:(.text._Z10loadConfigv+0x4c): undefined reference to `NidayandHelper::loadconfig()'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o:(.text._Z10loadConfigv+0x5c): undefined reference to `NidayandHelper::getconfig()'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o: In function `save<ArduinoJson::JsonVariant>':
> /Users/user/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Data/ValueSaver.hpp:35: undefined reference to `NidayandHelper::saveconfig(ArduinoJson::JsonVariant)'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o: In function `operator=<long int>':
> /Users/user/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObjectSubscript.hpp:41: undefined reference to `NidayandHelper::saveconfig(ArduinoJson::JsonVariant)'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o:(.text.setup+0xfc): undefined reference to `NidayandHelper::mqtt_gettopic(String)'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o:(.text.setup+0x154): undefined reference to `NidayandHelper::mqtt_gettopic(String)'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o:(.text.setup+0x183): undefined reference to `NidayandHelper::mqtt_gettopic(String)'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o: In function `setup':
> /Users/user/Downloads/motor-on-roller-blind-ws-master/motor_on_a_roller_blind-ws/motor_on_a_roller_blind-ws.ino:336: undefined reference to `NidayandHelper::mqtt_reconnect(PubSubClient&, String, String, std::list<char const*, std::allocator<char const*> >)'
> /Users/user/Downloads/motor-on-roller-blind-ws-master/motor_on_a_roller_blind-ws/motor_on_a_roller_blind-ws.ino:340: undefined reference to `NidayandHelper::mqtt_reconnect(PubSubClient&, String, String, std::list<char const*, std::allocator<char const*> >)'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o: In function `loop':
> /Users/user/Downloads/motor-on-roller-blind-ws-master/motor_on_a_roller_blind-ws/motor_on_a_roller_blind-ws.ino:379: undefined reference to `NidayandHelper::NidayandHelper()'
> sketch/motor_on_a_roller_blind-ws.ino.cpp.o: In function `~function':
> /Users/user/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional:2174: undefined reference to `NidayandHelper::NidayandHelper()'
> collect2: error: ld returned 1 exit status
> exit status 1
> Error compiling for board NodeMCU 1.0 (ESP-12E Module).

nidayand, 2018-06-25:
And you have installed the dependencies? E.g. PubSubClient

Kepete, 2018-07-03:
@nidayand  Yes, Stepper_28BYJ_48, PubSubClient, ArduinoJson, WIFIManager, WbSocketsServer are all installed.

Kepete, 2018-07-03:
Compiling some other examples(led blinking etc) works, so it isn't a connection or board error.

Kepete, 2018-07-03:
Problem solved. It didn't add the NidayandHelper.h file to the IDE, now compiled successfully. Issue can be closed.

closed 2018-07-03 by Kepete

fernandex00, 2018-12-01:
@nidayand hi i cant upload the code, got this error, i got all the files, what could it be
[image]