nidayand/motor-on-roller-blind-ws #14: not saving/loading current position on reboot

simo26246, 2018-07-17:
Hi there, first, let me say thank you for this. I've been looking for something similar for several months now. I had resigned myself to using servos, which was VERY hit/miss with respect to the position it opened to/stopped at plus, was an all or nothing (up/down).

Now that I've got this all up and running, when it loads back up, and I restart the web interface, it doesn't load its previously saved 'current position'. I query whether this is due to the fact that the mqtt messages aren't being retained? But honestly, have no idea about how to start looking at this!

This wouldn't be a problem, but for the fact that I kill the power to them when they are down (ie - at MAX). If I killed it when they're up, it saves the relative position fine and once I put it up, it goes down fine, but want it to think it starts at 100, rather than 0. 

Hope that makes sense - 

Here's a copy of the code from serial monitor:
```
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.94
MDNS responder started
Connect to http://Blind2.0_test.local or http://192.168.1.94
Registering MQTT server
Attempting MQTT connection...connected
Trying to send msg.../raw/esp8266/register:{ "id": "1386893", "ip":"192.168.1.94"}
Subscribed to /raw/esp8266/1386893/in
[0] get Text: (update)
Trying to send msg.../raw/esp8266/1386893/out:{ "set":0, "position":0 }
[0] get Text: 100
Trying to send msg.../raw/esp8266/1386893/out:{ "set":100, "position":0 }
Trying to send msg.../raw/esp8266/1386893/out:{ "set":100, "position":100 }
Stopped. Reached wanted position
Saved JSON to SPIFFS
{"currentPosition":5876,"maxPosition":5876,"config_name":"Blind2.0_test","mqtt_server":"hidden","mqtt_port":"1883","mqtt_uid":"hidden","mqtt_pwd":"hidden","config_rotation":}

```
then I kill power, then this pops back up:
```

*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.94
MDNS responder started
Connect to http://Blind2.0_test.local or http://192.168.1.94
Registering MQTT server
Attempting MQTT connection...connected
Trying to send msg.../raw/esp8266/register:{ "id": "1386893", "ip":"192.168.1.94"}
Subscribed to /raw/esp8266/1386893/in
[0] get Text: (update)
Trying to send msg.../raw/esp8266/1386893/out:{ "set":0, "position":0 }
```

Any help/guidance would be appreciated.

PaulHam211, 2018-08-09:
Any luck with this?

simo26246, 2018-08-10:
Nah, I gave up - couldn't figure out how to get it to send it with a retain flag. In addition, I couldn't get home assistant to extract the position value, rather than set - even parsing json in a template. 

What I did to get around it is change the orientation of up and down in home assistant, so at night when the blinds are down (which the program thinks is up) I can turn the power off. That way, at least it saves power for a bit of it, then, when it relaods, it thinks its up so will go down without having to recalibrate. I've got solar panels too, so using electricity in the day isn't such a concern for me. 

Sorry I couldn't be more help.

nidayand, 2018-08-27:
Hi, sorry for the lack for response. I'll take a look at it

simo26246, 2018-09-03:
Much appreciated. Will be interested to see your solution!

marecl, 2018-09-15:
I fixed it in my pull request about half a year ago. Please look there.

PaulHam211, 2018-11-07:
@marecl  Thank you, your PR worked a treat!